filipmu/Marlin-Mackerel #7: LCD 12864 - FullGraphics

designermx2, 2016-10-09:
When activate a 12864 display, there are a lot of errors.

will you add this display support too?

In file included from ultralcd.cpp:38:
dogm_lcd_implementation.h: In function 'void lcd_implementation_status_screen()':
dogm_lcd_implementation.h:179: error: 'fanSpeed' was not declared in this scope
ultralcd.cpp: In function 'void lcd_tune_menu()':
ultralcd.cpp:517: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:517: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:518: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:518: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:530: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:530: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:532: error: 'lcd_implementation_drawmenu_setting_edit_float6_selected' was not declared in this scope
ultralcd.cpp:532: error: 'lcd_implementation_drawmenu_setting_edit_float6' was not declared in this scope
ultralcd.cpp: In function 'void lcd_prepare_menu()':
ultralcd.cpp:707: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:707: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:708: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:708: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:710: error: 'lcd_implementation_drawmenu_setting_edit_float22_selected' was not declared in this scope
ultralcd.cpp:710: error: 'lcd_implementation_drawmenu_setting_edit_float22' was not declared in this scope
ultralcd.cpp:712: error: 'lcd_implementation_drawmenu_setting_edit_float6_selected' was not declared in this scope
ultralcd.cpp:712: error: 'lcd_implementation_drawmenu_setting_edit_float6' was not declared in this scope
ultralcd.cpp: In function 'void lcd_control_Filament_PID_menu()':
ultralcd.cpp:977: error: 'lcd_implementation_drawmenu_setting_edit_float6_selected' was not declared in this scope
ultralcd.cpp:977: error: 'lcd_implementation_drawmenu_setting_edit_float6' was not declared in this scope
ultralcd.cpp:978: error: 'lcd_implementation_drawmenu_setting_edit_float53_selected' was not declared in this scope
ultralcd.cpp:978: error: 'lcd_implementation_drawmenu_setting_edit_float53' was not declared in this scope
ultralcd.cpp:979: error: 'lcd_implementation_drawmenu_setting_edit_float53_selected' was not declared in this scope
ultralcd.cpp:979: error: 'lcd_implementation_drawmenu_setting_edit_float53' was not declared in this scope
ultralcd.cpp:980: error: 'lcd_implementation_drawmenu_setting_edit_float53_selected' was not declared in this scope
ultralcd.cpp:980: error: 'lcd_implementation_drawmenu_setting_edit_float53' was not declared in this scope
ultralcd.cpp: In function 'void lcd_init()':
ultralcd.cpp:1361: error: 'lcd' was not declared in this scope

Tartama, 2017-07-03:
Please, anyone have an update on this issue? Happened to me too

pcollingwood, 2019-06-03:
As far as I can see they have stated that it works with the 2004 (20 characters, 4 lines) not the full graphics controller 12864 (128x64 pixels). The firmware seems to be adapted from the Marlin offering, so the 12864 is an option there, but when using Marlin in a 3D printer there are other places that have to be changed too when using the 12864 (from memory - I've moved to Smoothie for my 3D printers).
I'm not involved in developing this system, but would suggest that you could just use a 2004 as reccomended in the documentation.

filipmu, 2019-11-10:
Since it works with the Char LCD this is more of a feature request than an issue.

bazz-77, 2020-03-08:
bought a  full graphic smart controller from reprapdiscount cannot load firmware nothing displaying on the LCD

killacycle, 2020-06-20:
I realize that the 12964 LCD is a major pain in the butt to implement, but it would really be appreciated. 

I have ordered a 2004 LCD controller, and it is on the way, so I actually can run this software, but it will take ~2 weeks (or more) to ship here to NZ. I looked to buy one locally, but could not find any local vendor with one in stock. <sigh> Not nearly as common as the 12864 FullGraphicSmartController. These 12864 LCD's typically come bundled with a RAMPS, and I have several on hand. I also have one I savaged from a dead 3D printer. I can buy the 12864 controller locally as well. 

Looking at the code and the documentation, this is _exactly_ the software my DIY filament extruder system needs. I've compiled it and loaded it, but my 12864 display is not implemented, so I can't actually see what it is doing or control what it is doing. :-( 

Perhaps a link that shows how to hack a 2004 LCD onto a 12864 control card might be helpful.

Hoping you will implement the 12864 LCD graphics controller someday. 

Bill D.

Mylan719, 2022-05-27:
Hi, I implemented and tested the 12864  Full Graphics controller support.

ellencbrk, 2022-05-30:
Thanks so much! That solved the problem I was having the past two weeks!

Mylan719, 2022-06-11:
- Added fix for strangely behaving encoder. It added numbers by increment of 4 per one "step".
- Fixed owerflowing text on 2 places
- settings menu looked diferently from all others, fixed that.