filipmu/Marlin-Mackerel #9: error uploading

Geurtswebdesign, 2016-12-14:
Hi, i downloaded the files but when i try to upload it i get a error
and it stops uploading could you help fix it??

regards Danny Geurts

Arduino: 1.6.13 (Windows 10), Board:"Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Mackerel.h:23:0,

                 from sketch\thermistortables.h:4,

                 from sketch\Configuration.h:847,

                 from D:\Downloads\Marlin-Mackerel-master\Marlin-Mackerel-master\Mackerel\Mackerel.ino:33:

sketch\pins.h:499:0: warning: "X_MAX_PIN" redefined

       #define X_MAX_PIN -1 //FMM remove definition for limit pins to speed the stepper interrupt.

 ^

sketch\pins.h:445:0: note: this is the location of the previous definition

     #define X_MAX_PIN           2

 ^

sketch\pins.h:500:0: warning: "Y_MAX_PIN" redefined

    #define Y_MAX_PIN -1

 ^

sketch\pins.h:451:0: note: this is the location of the previous definition

     #define Y_MAX_PIN          15

 ^

sketch\pins.h:501:0: warning: "Z_MAX_PIN" redefined

       #define Z_MAX_PIN -1

 ^

sketch\pins.h:457:0: note: this is the location of the previous definition

     #define Z_MAX_PIN          19

 ^

sketch\pins.h:502:0: warning: "X_MIN_PIN" redefined

       #define X_MIN_PIN -1

 ^

sketch\pins.h:444:0: note: this is the location of the previous definition

     #define X_MIN_PIN           3

 ^

sketch\pins.h:503:0: warning: "Y_MIN_PIN" redefined

       #define Y_MIN_PIN -1

 ^

sketch\pins.h:450:0: note: this is the location of the previous definition

     #define Y_MIN_PIN          14

 ^

sketch\pins.h:504:0: warning: "Z_MIN_PIN" redefined

       #define Z_MIN_PIN -1

 ^

sketch\pins.h:456:0: note: this is the location of the previous definition

     #define Z_MIN_PIN          18

 ^

In file included from sketch\Mackerel.h:23:0,

                 from sketch\Mackerel_main.cpp:36:

sketch\pins.h:499:0: warning: "X_MAX_PIN" redefined

       #define X_MAX_PIN -1 //FMM remove definition for limit pins to speed the stepper interrupt.

 ^

sketch\pins.h:445:0: note: this is the location of the previous definition

     #define X_MAX_PIN           2

 ^

sketch\pins.h:500:0: warning: "Y_MAX_PIN" redefined

    #define Y_MAX_PIN -1

 ^

sketch\pins.h:451:0: note: this is the location of the previous definition

     #define Y_MAX_PIN          15

 ^

sketch\pins.h:501:0: warning: "Z_MAX_PIN" redefined

       #define Z_MAX_PIN -1

 ^

sketch\pins.h:457:0: note: this is the location of the previous definition

     #define Z_MAX_PIN          19

 ^

sketch\pins.h:502:0: warning: "X_MIN_PIN" redefined

       #define X_MIN_PIN -1

 ^

sketch\pins.h:444:0: note: this is the location of the previous definition

     #define X_MIN_PIN           3

 ^

sketch\pins.h:503:0: warning: "Y_MIN_PIN" redefined

       #define Y_MIN_PIN -1

 ^

sketch\pins.h:450:0: note: this is the location of the previous definition

     #define Y_MIN_PIN          14

 ^

sketch\pins.h:504:0: warning: "Z_MIN_PIN" redefined

       #define Z_MIN_PIN -1

 ^

sketch\pins.h:456:0: note: this is the location of the previous definition

     #define Z_MIN_PIN          18

 ^

sketch\Mackerel_main.cpp:2408:36: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

         LCD_MESSAGEPGM(MACHINE_NAME" "MSG_OFF".");

                                    ^

In file included from sketch\SdFile.h:27:0,

                 from sketch\cardreader.h:8,

                 from sketch\Mackerel_main.cpp:50:

SdBaseFile.h:38: error: using typedef-name 'fpos_t' after 'struct'

 struct fpos_t {

        ^

In file included from sketch\Mackerel.h:10:0,

                 from sketch\Mackerel_main.cpp:36:

c:\program files (x86)\arduino\hardware\tools\avr\avr\include\stdio.h:950:33: note: 'fpos_t' has a previous declaration here

 __extension__ typedef long long fpos_t;

                                 ^

exit status 1
using typedef-name 'fpos_t' after 'struct'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

filipmu, 2016-12-16:
This error ' using typedef-name 'fpos_t' after 'struct'   ' looks like its caused by a change in a later arduino version.  This compiles under arduino 1.0.6.  See this post about Marlin with someone with the same issue:
http://forums.reprap.org/read.php?415,709213,709213

The post suggests a fix.  I will take a look and see if I can make the fix.

filipmu, 2016-12-16:
This is fixed.  Changed fpos_t to fatpos_t in the two files SdBaseFile.h and SdBaseFile.cpp.  The error is caused by the fact that fpos_t is a reserved word in arduino1.16.3 but was not reserved in an older version of arduino.

closed 2016-12-16 by filipmu