filipmu/Marlin-Mackerel #10: Adapt firmware

santisantisanti, 2018-03-01:
Hi Filipmu!

I bought a machine to make filament but it is not performing as good as I would like, so I decided I will add a filament sensor and connect it, through ramps 1.4, to the puller, displaying all necessary information in a 2004 LCD (RPM of the puller motor - measured diameter of filament in mm - filament feed rate at puller in mm/sec - lenght of filament produced in mm - average measured filament diameter - minimum measured filament diameter - maximum measured filament diameter).

For what I am seeing, it is contemplated in the firmware, but I want it to be an isolated system, that only controls the puller speed due to the real time readings of filament diameter, because the other parts of the machine run ok and I don't want to change them because I don't have the current firmware. The seller is not answering my mails and isn't giving me support as he promised.

Can you help me a bit telling me what do I have to modify to achieve this? I am really frustrated with the machine I have and I think this is the solution.

I would appretiate it a lot.

Thank you and greetings from Argentina!

Santiago Serrano.

pcollingwood, 2019-06-03:
I'm doing the same thing with a Filastruder. I have issues with just how slowly it extrudes, so want to go to a much bigger nozzle (5mm) and use the centre section of this - just the measuring system and puller. What I'm going to try is to add things like the thermistor and step bot for the extruder, have the thermistor on the barel so it get's readings (so that it doesn't shut it down) and see if that works. I'll play with the firmware to see what can be commented out later. ATM I think all the neccessary data is being displayed so it should work. I'll post here if it does.
BTW - Thanks guys for a GREAT system. If I hadn't already bought a Filastruder I'd have made this system fron the start.

filipmu, 2019-11-10:
Since you are not using the temp control or extruder motor control, don't wire the extruder motor or temp sensor up.  Then in config you can config the extruder (and puller) motor to run, even tho the temp is too low.  In the UI you should then be able to set the puller speed, and turn on PID control etc.